HelioScan/HelioScan #292: Intrinsic imaging: generic component error

phaedrus2014, 2015-02-17:
Hello.

I'm having trouble with using HellioScan for intrinsic imaging.
When I launch the HelioScan_3.vi from LabVIEW project browser, it doesn't start with the message showing
GenericComponent: create_instance.vi:
configuration with this label not found

[image]
 I configured camera mode and intrinsic optical imaging mode as written in the user`s guide before launching HelioScan_3.vi. I have no idea which configuration is missing.

My current system is as follows

A.windows7 64bit
B.LabVIEW 2010 SP1 32bit
C.NI Vision development module (2013)
D.NI Device drivers (2013 Feb)
E.EasyXML(full version)
F.VI package manager(2014)

I will appreciate your help with this situation.

phaedrus2014, 2015-02-17:
my current configuration

[image]

mtedaldi, 2015-02-17:
Please also provide the error log which is stored in your user directory under 
"Documents\LabVIEW Data\helioscan\errors"
Also:
What Version of Helioscan?
Did Helioscan already run with that configuration?
How was this configuration created?

phaedrus2014, 2015-02-17:
thanks for the reply.

here is the captured image of error log
[image]

What Version of Helioscan? →　version of the Helioscan is 3.10.0
Did Helioscan already run with that configuration?　→ not yet.
How was this configuration created?　→　Configuration for camera and intrinsic imaging were created by using configuration wizard runner ( runner.vi )

[image]

phaedrus2014, 2015-02-17:
Thanks for the reply

here is the error log
[image]

What Version of Helioscan?  →　version is 3.10.0
 Did Helioscan already run with that configuration? →　not yet
 How was this configuration created?　→　Configuration for camera and intrinsic imaging were created by using wizard runner.( runner.vi )
[image]
[image]

mtedaldi, 2015-03-12:
Did you also do "main configuration"? It definitely look list not everything is configured so helioscan does not know which modules to load.

phaedrus2014, 2015-03-31:
Sorry for late response. 
When I asked the previous question, "main configuration" for camera mode was done as written in the manual. But, I later found that configuration for objective was left blank.
Anyway, I could fix by myself and "Camera mode" seems working now.

BUT, intrinsic optical imaging mode still doesn't work.
As shown below, progress bar stops halfway during loading with no error message.
[image]

Here is my current configuration.
[image]

I'm still not sure about how I can properly configure the stimulator TLC, and I now suspect this is the problem.
I have checked the manual several times, so other configurations (including main configuration) are probably correctly done as described in the manual. 

As a stimulator TLC I added StimulatorDL090727Writer using configuration manager and configured as follows.
 SignalWriter: "signalwriterdl090202DAQmx.lvclass" with configuration "intrinsic"
Timing clock: ClockDL100201DAQmx.lvclass with configuration *intrinsic" and sampling rate 10Hz.

I also specified stimulator TLC(StimulatorDL090727writer) in the stimulator tab of the ImagingMODEMG091111Intrinsicimaging. 
[image]

My DAQ is NI usb-6259 BNC.
I will appreciate your help with this situation.